Mauin/RxFingerprint #8: Problems with orientation changes during listening

mzander, 2016-10-09:
Hi Mauin,

do you experience problems on changing the orientation while listening?

I am calling RxFingerprint.authenticate(this) in onResume on my activity and store the subscription in a member variable.
Within the onPause method I unsubsribe my fingerprintObersable by fingerprintObersable.unsubscribe and set the reference to null.

I get this Exception in catlog as debug info:

```
Fingerprint operation canceled.
at com.mtramin.rxfingerprint.FingerprintObservable$1.onAuthenticationError(FingerprintObservable.java:76)
at android.support.v4.hardware.fingerprint.FingerprintManagerCompat$Api23FingerprintManagerCompatImpl$1.onAuthenticationError(FingerprintManagerCompat.java:290)
at android.support.v4.hardware.fingerprint.FingerprintManagerCompatApi23$1.onAuthenticationError(FingerprintManagerCompatApi23.java:86)
at android.hardware.fingerprint.FingerprintManager$MyHandler.sendErrorResult(FingerprintManager.java:721)
at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:695)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:148)
at android.app.ActivityThread.main(ActivityThread.java:5417)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
```

and those debug messages in logcat:

```
10-09 20:53:03.673 1263-1263/? V/fingerprintd: authenticate(sid=0, gid=0)
10-09 20:53:09.774 1263-1263/? V/fingerprintd: stopAuthentication()
10-09 20:53:09.774 1263-1263/? D/FingerprintHal: ----------------> fingerprint_cancel ----------------->
10-09 20:53:09.774 1263-1263/? D/fingerprintd: onError(5)
10-09 20:53:09.778 1263-1263/? V/fingerprintd: authenticate(sid=0, gid=0)
10-09 20:53:18.757 1263-1263/? V/fingerprintd: authenticate(sid=0, gid=0)
10-09 20:55:36.335 1568-1583/system_process I/ActivityManager: Config changes=480 {1.0 310mcc260mnc en_US ldltr sw411dp w689dp h387dp 420dpi nrml land finger -keyb/v/h -nav/h s.7}
10-09 20:55:36.377 1263-1263/? V/fingerprintd: stopAuthentication()
10-09 20:55:36.377 1263-1263/? D/FingerprintHal: ----------------> fingerprint_cancel ----------------->
10-09 20:55:36.377 1263-1263/? D/fingerprintd: onError(5)
10-09 20:55:36.429 1263-1263/? V/fingerprintd: authenticate(sid=0, gid=0)
10-09 20:55:36.432 1263-1263/? V/fingerprintd: stopAuthentication()
10-09 20:55:36.432 1263-1263/? D/FingerprintHal: ----------------> fingerprint_cancel ----------------->
10-09 20:55:36.432 1263-1263/? D/fingerprintd: onError(5)
```

It would be great if you can help me keep using this library. I just considered trying an other one but I really love the Rx'ness of yours 👍 

Thanks,
Moritz

mzander, 2016-10-09:
The weird thing is, that is only happens if I start the device in portrait mode and change the orientation to landscape.
If I start the activity in landscape and change the orientation to portrait. **it still works after the orientation change**

Mauin, 2016-10-10:
Just for completeness, which version of RxFingerprint are you using? I'll give this one a look and try to reproduce this.

Mauin, 2016-10-10:
I'm able to reproduce this issue. Thanks for reporting. I'll see if it makes sense to supress the Exception, because it is kind of wanted behaviour to cancel the Fingerprint operation when the Observable gets unsubscribed from. This currently throws the error.

mzander, 2016-10-10:
Thank you for investigating.
But why the re-registering doesnt work anymore?

Mauin, 2016-10-11:
I found a way to work around this in the library and will open a PR for this shortly. Good find on this @mzander. Possibly there are also issues in the Android SDK that lead to this behaviour, I'll investigate a bit more.

Mauin, 2016-10-13:
I just merged #13 which should fix this. I'll release a new version of RxFingerprint shortly. Would be awesome if you can give me feedback on the new version if this fixes your issue. I wasn't able to reproduce the behaviour you described anymore with the fixes.

closed 2016-10-13 by Mauin

mzander, 2016-10-13:
Hey Mauin,

great thank you.
I will test version 1.2.0 as soon as possible and send you my feedback.

Thank you very much for your awesome support!
Moritz

PR #13 (2016-10-13): discourage use of application context
This will fix #8.

As described in the javadoc, the issue is that when using an application context for calling the FingerprintManager, the cancellation of the first operation will call the callback of the second operation. This will call onError on the emitter which will unsubscribe the authentication Observable, resulting in a cancelled authentication operation when the application changes orientation.