Mauin/RxFingerprint #9: crash on Samsung s5 with os marshmallow and aslo have sensor

MuhammadArsalanChishti, 2016-10-10:


Mauin, 2016-10-10:
I have no S5 here to reproduce this with, can you please post a stacktrace in here so that I can take a look?

MuhammadArsalanChishti, 2016-10-10:
Subscription subscription = RxFingerprint.authenticate(MainActivity.this)
            .subscribe(fingerprintAuthenticationResult -> {
              switch (fingerprintAuthenticationResult.getResult()) {
                case FAILED:
                  setStatusText("Fingerprint not recognized, try again!");
                  break;
                case HELP:
                  setStatusText(fingerprintAuthenticationResult.getMessage());
                  break;
                case AUTHENTICATED:
                  setStatusText("Successfully authenticated!");
                  break;
              }
            }, throwable -> {
              Log.e("ERROR", "authenticate", throwable);
            });
      }
    });

---------LOG------------

/ERROR: authenticate
                                                           java.lang.IllegalAccessException: Fingerprint authentication is not available on this device! Ensure that the device has a Fingerprint sensor and enrolled Fingerprints by calling RxFingerprint#available(Context) first
                                                               at com.mtramin.rxfingerprint.FingerprintObservable.call(FingerprintObservable.java:54)
                                                               at com.mtramin.rxfingerprint.FingerprintObservable.call(FingerprintObservable.java:37)
                                                               at rx.internal.operators.OnSubscribeFromAsync.call(OnSubscribeFromAsync.java:71)
                                                               at rx.internal.operators.OnSubscribeFromAsync.call(OnSubscribeFromAsync.java:31)
                                                               at rx.Observable.subscribe(Observable.java:10246)
                                                               at rx.Observable.subscribe(Observable.java:10213)
                                                               at rx.Observable.subscribe(Observable.java:10053)
                                                               at com.apifingerprint.MainActivity$1.onClick(MainActivity.java:76)
                                                               at android.view.View.performClick(View.java:5697)
                                                               at android.widget.TextView.performClick(TextView.java:10826)
                                                               at android.view.View$PerformClick.run(View.java:22526)
                                                               at android.os.Handler.handleCallback(Handler.java:739)
                                                               at android.os.Handler.dispatchMessage(Handler.java:95)
                                                               at android.os.Looper.loop(Looper.java:158)
                                                               at android.app.ActivityThread.main(ActivityThread.java:7224)
                                                               at java.lang.reflect.Method.invoke(Native Method)
                                                               at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                               at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
10-10 11:54:36.796 29330-29330/com.apifingerprint D/AndroidRuntime: Shutting down VM
10-10 11:54:36.796 29330-29330/com.apifingerprint E/AndroidRuntime: FATAL EXCEPTION: main
                                                                    Process: com.apifingerprint, PID: 29330
                                                                    java.lang.IncompatibleClassChangeError: The method 'void com.mtramin.rxfingerprint.FingerprintObservable.com_mtramin_rxfingerprint_FingerprintObservable_lambda$call$0()' was expected to be of type direct but instead was found to be of type virtual (declaration of 'com.mtramin.rxfingerprint.FingerprintObservable' appears in /data/app/com.apifingerprint-2/base.apk)
                                                                        at com.mtramin.rxfingerprint.FingerprintObservable.access$lambda$0(FingerprintObservable.java)
                                                                        at com.mtramin.rxfingerprint.FingerprintObservable$$Lambda$1.call(Unknown)
                                                                        at rx.subscriptions.BooleanSubscription.unsubscribe(BooleanSubscription.java:71)
                                                                        at rx.internal.subscriptions.SequentialSubscription.update(SequentialSubscription.java:75)
                                                                        at rx.subscriptions.SerialSubscription.set(SerialSubscription.java:50)
                                                                        at rx.internal.operators.OnSubscribeFromAsync$BaseAsyncEmitter.setSubscription(OnSubscribeFromAsync.java:175)
                                                                        at com.mtramin.rxfingerprint.FingerprintObservable.call(FingerprintObservable.java:62)
                                                                        at com.mtramin.rxfingerprint.FingerprintObservable.call(FingerprintObservable.java:37)
                                                                        at rx.internal.operators.OnSubscribeFromAsync.call(OnSubscribeFromAsync.java:71)
[... cut ...]
                                                                        at android.app.ActivityThread.main(ActivityThread.java:7224)
                                                                        at java.lang.reflect.Method.invoke(Native Method)
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)

Mauin, 2016-10-10:
Which result does the method `RxFingerprint.isAvailable(context)` return?

MuhammadArsalanChishti, 2016-10-10:
returns false

MuhammadArsalanChishti, 2016-10-10:
basically it is not  detection hardware

Mauin, 2016-10-10:
If you manually call `FingerprintManagerCompat.from(context).isHardwareDetected()` that also returns false?

MuhammadArsalanChishti, 2016-10-10:
yes

MuhammadArsalanChishti, 2016-10-10:
boolean rtn = RxFingerprint.isAvailable(MainActivity.this);
        if (!rtn) {
          Toast.makeText(getApplicationContext(), "Hardware not Detected", Toast.LENGTH_SHORT).show();
          return;
        } else {
          Subscription subscription = RxFingerprint.authenticate(getApplicationContext())
              .subscribe(fingerprintAuthenticationResult -> {
                switch (fingerprintAuthenticationResult.getResult()) {
                  case FAILED:
                    setStatusText("Fingerprint not recognized, try again!");
                    break;
                  case HELP:
                    setStatusText(fingerprintAuthenticationResult.getMessage());
                    break;
                  case AUTHENTICATED:
                    setStatusText("Successfully authenticated!");
                    break;
                }
              }, throwable -> {
                Log.e("ERROR", "authenticate", throwable);
              });
        }

MuhammadArsalanChishti, 2016-10-10:
the samsung galaxy s5 built in os was kitkat 4.4.2 and upgradable to v6.0 (Marshmallow).I think this was the issue?

Mauin, 2016-10-10:
Yes, that was definitely the issue. If it works now that's awesome. 

Pre-Marshmallow Samsung devices used some internal Samsung APIs and didn't support the Fingerprint APIs, Samsung devices with fingerprint sensors only support the official APIs with Marshmallow and above.

So it works for you now with Marshmallow?

MuhammadArsalanChishti, 2016-10-10:
no its not working.
it works on Samsung pass sdk for finger print.
ok . Its mean the  device with builtin os 6.0 will work?

MuhammadArsalanChishti, 2016-10-10:
also crashes in samsung s7 edge

MuhammadArsalanChishti, 2016-10-10:
NOT REGISTERED FINGER PRINT SAY NOT RECOGNIZED but when registered finger touches sensor it crashes
------LOG------------
FATAL EXCEPTION: main
                                                                    Process: com.apifingerprint, PID: 18722
                                                                    rx.exceptions.UnsubscribeFailedException: The method 'void com.mtramin.rxfingerprint.FingerprintObservable.com_mtramin_rxfingerprint_FingerprintObservable_lambda$call$0()' was expected to be of type direct but instead was found to be of type virtual (declaration of 'com.mtramin.rxfingerprint.FingerprintObservable' appears in /data/app/com.apifingerprint-2/base.apk)
                                                                        at rx.observers.SafeSubscriber.onCompleted(SafeSubscriber.java:93)
                                                                        at rx.internal.operators.OnSubscribeFromAsync$BaseAsyncEmitter.onCompleted(OnSubscribeFromAsync.java:128)
                                                                        at rx.internal.operators.OnSubscribeFromAsync$LatestAsyncEmitter.drain(OnSubscribeFromAsync.java:487)
                                                                        at rx.internal.operators.OnSubscribeFromAsync$LatestAsyncEmitter.onCompleted(OnSubscribeFromAsync.java:442)
                                                                        at com.mtramin.rxfingerprint.FingerprintAuthenticationObservable.onAuthenticationSucceeded(FingerprintAuthenticationObservable.java:61)
                                                                        at com.mtramin.rxfingerprint.FingerprintObservable$1.onAuthenticationSucceeded(FingerprintObservable.java:94)
                                                                        at android.support.v4.hardware.fingerprint.FingerprintManagerCompat$Api23FingerprintManagerCompatImpl$1.onAuthenticationSucceeded(FingerprintManagerCompat.java:301)
                                                                        at android.support.v4.hardware.fingerprint.FingerprintManagerCompatApi23$1.onAuthenticationSucceeded(FingerprintManagerCompatApi23.java:96)
                                                                        at android.hardware.fingerprint.FingerprintManager$MyHandler.sendAuthenticatedSucceeded(FingerprintManager.java:1108)
                                                                        at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:1052)
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                        at android.os.Looper.loop(Looper.java:158)
                                                                        at android.app.ActivityThread.main(ActivityThread.java:7229)
                                                                        at java.lang.reflect.Method.invoke(Native Method)
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
                                                                     Caused by: java.lang.IncompatibleClassChangeError: The method 'void com.mtramin.rxfingerprint.FingerprintObservable.com_mtramin_rxfingerprint_FingerprintObservable_lambda$call$0()' was expected to be of type direct but instead was found to be of type virtual (declaration of 'com.mtramin.rxfingerprint.FingerprintObservable' appears in /data/app/com.apifingerprint-2/base.apk)
                                                                        at com.mtramin.rxfingerprint.FingerprintObservable.access$lambda$0(FingerprintObservable.java)
                                                                        at com.mtramin.rxfingerprint.FingerprintObservable$$Lambda$1.call(Unknown)
                                                                        at rx.subscriptions.BooleanSubscription.unsubscribe(BooleanSubscription.java:71)
                                                                        at rx.internal.subscriptions.SequentialSubscription.unsubscribe(SequentialSubscription.java:180)
                                                                        at rx.subscriptions.SerialSubscription.unsubscribe(SerialSubscription.java:36)
                                                                        at rx.internal.operators.OnSubscribeFromAsync$BaseAsyncEmitter.unsubscribe(OnSubscribeFromAsync.java:148)
                                                                        at rx.internal.util.SubscriptionList.unsubscribeFromAll(SubscriptionList.java:136)
                                                                        at rx.internal.util.SubscriptionList.unsubscribe(SubscriptionList.java:125)
                                                                        at rx.Subscriber.unsubscribe(Subscriber.java:98)
                                                                        at rx.observers.SafeSubscriber.onCompleted(SafeSubscriber.java:90)
[... cut ...]
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                                        at android.os.Looper.loop(Looper.java:158) 
                                                                        at android.app.ActivityThread.main(ActivityThread.java:7229) 
                                                                        at java.lang.reflect.Method.invoke(Native Method)

Mauin, 2016-10-11:
In this last stacktrace the Fingerprint authentication is working fine. The sensor gets recognized and authentication is successful.

Which version of RxFingerprint are you using in that example?

MuhammadArsalanChishti, 2016-10-11:
yes i debug it and it falls on OnSuccess but crashes before showing toast

Mauin, 2016-10-11:
Do you use Retrolambda in your example? 
I've read some similar issues that were tied to Retrolambda which I'm using in this lib.

Mauin, 2016-10-19:
Have you been able to find more information on the issue in the last Stacktrace you posted? Have you tried using the latest version of RxFingerprint?

MuhammadArsalanChishti, 2016-10-20:
i have done it without your library

Mauin, 2016-10-20:
Okay, then I will close this one.

closed 2016-10-20 by Mauin

MuhammadArsalanChishti, 2016-10-20:
yes (y)