Mauin/RxFingerprint #15: More specific error message in RxFingerprint.decrypt

CDehning, 2016-10-19:
Hey there,

Having some issues with RxFingerprint.decrypt here: In very rare cases this crashes for some of our users (mostly on Samsung S5 and S6) with the error message "Invalid input given for decryption operation. Make sure you provide a string that was previously encrypted by RxFingerprint.".

Can you extend this error message with something telling us if the encrypted string was empty or if it was missing the separator? Might help tracking down the issue.

The error message comes from CryptoData:53 and could be something like this:
"Invalid input given for decryption operation. Make sure you provide a string that was previously encrypted by RxFingerprint. encryptedData.length: 0, contains separator: false”

Thanks!

Mauin, 2016-10-19:
Good idea, I'll enhance the error message. Might also add some additional checks to the encryption to verify the encrypted value has the correct format before returning it to the caller.

Mauin, 2016-10-20:
v1.2.2 was just released which adds the more detailed error messages.

closed 2016-10-20 by Mauin