Mauin/RxFingerprint #20: Lib crashes on unlock

Andoctorey, 2016-11-23:
`            com.mtramin.rxfingerprint.data.FingerprintAuthenticationException: Fingerprint operation canceled.
                                                                                  at com.mtramin.rxfingerprint.FingerprintObservable$1.onAuthenticationError(FingerprintObservable.java:83)
                                                                                  at android.support.v4.hardware.fingerprint.FingerprintManagerCompat$Api23FingerprintManagerCompatImpl$1.onAuthenticationError(FingerprintManagerCompat.java:290)
                                                                                  at android.support.v4.hardware.fingerprint.FingerprintManagerCompatApi23$1.onAuthenticationError(FingerprintManagerCompatApi23.java:90)
                                                                                  at android.hardware.fingerprint.FingerprintManager$MyHandler.sendErrorResult(FingerprintManager.java:721)
                                                                                  at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:695)
                                                                                  at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                  at android.os.Looper.loop(Looper.java:148)
                                                                                  at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                                                  at java.lang.reflect.Method.invoke(Native Method)
                                                                                  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                                                  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)`

Mauin, 2016-11-23:
From the stacktrace it seems like the Fingerprint operation was cancelled by unsubscribing the Observable. Could you share a bit more details in which condition this happens?

Also: Are you handling the `onError` callback of the RxFingerprint `Observable`?

Andoctorey, 2016-11-23:
Thanks for fast response. It seems already found problem, forgot about disposable.dispose().

Mauin, 2016-11-23:
Perfect, glad you found it! Can I close this issue then?

Andoctorey, 2016-11-23:
It seems it is still crashing when I minimize app: https://gist.github.com/Andoctorey/d7d3673f2db9ad3f3987cf0910196267

Mauin, 2016-11-23:
Thanks for the gist. I'll take a look at it and I'll try to reproduce this to see what's causing this. Once you're calling disposed, the error shouldn't be emitted anymore.

Andoctorey, 2016-11-23:
Maybe it will help.  If I just call rxDisposable.dispose() after subscription - it crashes.

Mauin, 2016-11-23:
Just reproduced and verified this on my end. This should affect v2.x only. I'll get started on a fix for this one.

Mauin, 2016-11-23:
I'll publish a new release shortly that will address this.

closed 2016-11-23 by Mauin

Mauin, 2016-11-23:
@Andoctorey I just released v2.0.1 of the library. This should fix the issue. Thanks for the report!

Andoctorey, 2016-11-24:
Thanks, it is fixed!

PR #21 (2016-11-23): prevent onError to be emitted when emitter is disposed
Fixes #20 

This prevents onErrors to be emitted when `FingerprintObservable`s have already been disposed of.
This caused issues when the `FingerprintObservable` was disposed of, since this cancels the Fingerprint operation with a `CancellationSignal` internally. The Android Fingerprint API would then callback with `onAuthenticationError` "Fingerprint operation canceled." which would actually throw because of how RxJava2 works internally.