Mauin/RxFingerprint #23: Recover from Fingerprint operation canceled

mikeKickserv, 2016-11-23:
Hey,

This could be the fact that I am new to RxJava but when I cancel the fingerprint I get the exception com.mtramin.rxfingerprint.data.FingerprintAuthenticationException: Fingerprint operation canceled. that I cannot seem to fail gracefully with.  I have tried putting error handlers in two places, neither of which seem to be working correctly. Below is my code

`fingerprintDisposable = RxFingerprint.authenticate(this)
                .onErrorReturn(new Function<Throwable, FingerprintAuthenticationResult>() {
                    @Override
                    public FingerprintAuthenticationResult apply(Throwable throwable) throws Exception {
                        try {
                            return new FingerprintAuthenticationResult(FingerprintResult.FAILED, null);
                        } catch(Exception e) {
                            Log.e("ERROR", e.getLocalizedMessage());
                            return new FingerprintAuthenticationResult(FingerprintResult.FAILED, null);
                        }
                    }
                })
                .subscribe(new Consumer<FingerprintAuthenticationResult>() {
                    @Override
                    public void accept(FingerprintAuthenticationResult fingerprintAuthenticationResult) throws Exception {
                        switch (fingerprintAuthenticationResult.getResult()) {
                            case FAILED:
                                Toast.makeText(SignInActivity.this, R.string.fingerprint_not_recognized, Toast.LENGTH_LONG);
                                break;
                            case HELP:
                                Toast.makeText(SignInActivity.this, fingerprintAuthenticationResult.getMessage(), Toast.LENGTH_LONG);
                                break;
                            case AUTHENTICATED:
                                ((ImageView) dialog.findViewById(R.id.fingerprint_icon)).setImageResource(R.drawable.fingerprint_ok);
                                ((TextView) dialog.findViewById(R.id.touch_sensor_tv)).setText(R.string.fingerprint_recognized);

                                onSignIn(mSessionService.getFingerprintUsername(), mSessionService.getFingerprintPassword());
                                break;
                        }
                    }
                }, new Consumer<Throwable>() {
                    @Override
                    public void accept(Throwable throwable) throws Exception {
                        Log.e("ERROR", "", throwable);
                    }
                });`

Mauin, 2016-11-23:
Thanks for the report. This sounds similar to the issue in #20. Your code looks fine so far.

I'll publish a new version of the library shortly to prevent this from happening!

Mauin, 2016-11-23:
I just released v2.0.1 of the library. That should fix this issue. In that case the `.doOnError()` block in your code should also not be necessary anymore.

Thanks for the report! Let me know if you face any further issues!

mikeKickserv, 2016-11-23:
Thank you very much for the prompt reply! I will look into it Monday.

Andoctorey, 2016-11-24:
Thanks, it is fixed on v2.0.1

Mauin, 2016-11-29:
@mikeKickserv Did it work out okay for you?

Mauin, 2016-12-01:
I'm closing this because of inactivity, feel free to re-open this one or create a new issue if you find out more.

closed 2016-12-01 by Mauin

mikeKickserv, 2016-12-01:
Never got back to u. Fix was great! thanks