Mauin/RxFingerprint #24: KeyPermanentlyInvalidatedException on encrypt()

jrmybrault, 2016-12-08:
Hey,

I'm using RxFingerprint (1.2.2) encryption / decryption capabilities and I'm currently struggling with the case where key is invalidated. Here is the flow of what I'm doing :

1) Launch the activity => Ask user for some data => Encrypt that data => Save the encrypted data
2) Remove the fingerprint used for the previous encryption, then add a new one
3) Re-launch the activity => Decrypt the data => KeyPermanentlyInvalidatedException (that's fair) => Erase the encrypted data
4) Re-ask user for some data => Re-encrypt that data => KeyPermanentlyInvalidatedException (that does not seems right ?)

So it seems that I can't **re**-use encryption at all after a KeyPermanentlyInvalidatedException. Am I missing something and if not what could be the cause of this result ?

Mauin, 2016-12-08:
Thanks for reporting this. I'll take a look into the issue and see if I can reproduce this with the steps you mentioned. 

But overall you're right. As per the Android API documentation the KeyInvalidatedException should only occur once when the user changes his fingerprint settings on the device. 

Just for completness sake: Which devices did you see this on?

JulienDev, 2016-12-09:
Hi Mauin, the issue occurs on Nexus 5x and Nexus 6p. We didn't try on other devices.

Mauin, 2016-12-10:
@jrmybrault @JulienDev The Android APIs actually behave correctly. They invalidate the key, which RxFingerprint generates automatically behind the curtains if you don't explicitly state which key to use. However, RxFingerprint only ever generates one Key. And once that one gets invalidated due to changes in the users fingerprint setup the auto-generated key is unusable for further crypto operations.

I'm thinking if/how I can improve this situation, but might end up deprecating the calls to `encrypt`/`decrypt` so that users of RxFingerprint themselves have to specify a `keyName` and change that key once it was invalidated.

jrmybrault, 2016-12-27:
@Mauin : alright that makes sense, deprecation or a least a warning in the Javadoc might be wise then. Thanks for the explanation !

GavinPacini, 2017-01-26:
Is there anyway to remove the key from the keystore? 

Because, I think a nice way to use the API would be to try encrypt and then in onNext have a new `FingerprintResult` which gives you the option to delete the key and create a new one (if `RxFingerprint` detects a `KeyPermanentlyInvalidatedException`.

Something like that perhaps.

Mauin, 2017-01-26:
@GavinPacini So in the error case for a `KeyPermanentlyInvalidatedException` having the possibility to simply delete the key, and reuse the same keyName (which will then be a new key) for the next encryption?

Sounds like a viable solution if it works. Feel free to try it out and submit a PR!

Mauin, 2017-02-08:
Just merged the PR that will delete and renew the key in `RxFingerprint.encrypt()` if it has been invalidated before.

PR #34 (2017-02-08): Renew Key on KeyPermanentlyInvalidatedException
Resolves #24 

This will delete and renew the key in the Keystore when calling `RxFingerprint.encrypt(...)` when the  keys were invalidated.
closed 2017-02-08 by Mauin