Mauin/RxFingerprint #26: How to use encryption logic

nksaroj, 2017-01-24:
I am getting the token from the backend I am looking to encrypt it using this library and store it in SharedPrefrence and on subsequent Logon I would like to read the encrypted token from the shared preference and decrypt using the Fingerprint and send the decrypted token to backend 
It would be great if you could show some lights

Mauin, 2017-01-24:
In your given example this would be done by calling `RxFingerprint.encrypt(context, "your_key_name", <token you received from your backend>)`. The user has to then touch the fingerprint sensor and you should display some UI that tells him to do so.

In the `onNext` of the resulting subscription you will then receive a result from which you can retrieve the encrypted value. You can store this value in the SharedPreferences.

When you later want to decrypt this value you call `RxFingerprint.decrypt(context, "your_key_name", <encrypted token from SharedPreference>)` which the user has to again confirm with his fingerprint. This will return you the decrypted token which you can then use for your backend.

You can see more detailed code examples in the sample.

nksaroj, 2017-01-24:
Thanks mate. Can I able to generate the encrypted token without asking the user to touch on fingerprint during enablement process?

Mauin, 2017-01-24:
Not with the way RxFingerprint is currently set up. I might figure out a way to add this in the future.

petrdvorak, 2017-02-04:
@Mauin IMO this is not possible on Android. Unlike on iOS, where you can store keys to Keychain without Touch ID and only retrieval requires Touch ID, the key generation on Android requires fingerprint authentication to generate the encryption key.

When I was trying to accomplish this, the `SecretKey` object was not properly initialized when I was calling `keyGenerator.generateKey()` without prior fingerprint authentication...

AKiniyalocts, 2017-03-02:
I am facing a related issue. Once fingerprint authentication is enabled in our application, we request a new cookie from the server, that cookie is then encrypted via 

`RxFingerprint.encrypt(this, KEY, loginCookie)`

That encrypted cookie is then persisted.

Then, when authentication is needed because of cookie expiration, user authentication is required and the cookie is decrypted via:

`RxFingerprint.decrypt(this, KEY, encryptedCookie)`

Once decrypted, we perform our login request with this unencrypted cookie. Upon a response from that login, a new login cookie is returned, so we prompt the user again for authentication to encrypt the cookie via:

`RxFingerprint.encrypt(this, KEY, loginCookie)`

I'm wondering if we can leverage [setUserAuthenticationValidityDurationSeconds()](https://developer.android.com/reference/android/security/keystore/KeyGenParameterSpec.Builder.html#setUserAuthenticationValidityDurationSeconds(int)) inside of `KeyGenParameterSpec.Builder`
 to not require fingerprint authentication from the user if each encryption/decryption request falls within that set window.

Mauin, 2017-03-02:
@AKiniyalocts That sounds interesting. I'd definitely be up for opening up the `KeyGenParameterSpec` a bit to make use cases like yours possible. 

I'll take a closer look at it on the weekend if I find a way to make this easily possible without changing the API of the library too much.

PR #50 (2017-04-25): Add support for RSA encryption
This PR adds the possibility to use fingerprint authentication with RSA public-private key authentication. Doing so allows to encrypt values without fingerprint authentication that can then later only be decrypted with fingerprint authentication.

Numerous things were refactored for this PR to allow for both the new RSA crypto and the previous AES crypto to work side by side. 

RxFingerprint gains a two new methods:
- `RxFingerprint.encrypt(EncryptionMethod, Context, String, String)`
- `RxFingerprint.decrypt(EncryptionMethod, Context, String, String)`

`Encryptio [... cut]
closed 2017-04-25 by Mauin