Mauin/RxFingerprint #27: Proguard Error

ragdroid, 2017-01-26:
I am using version `1.2.2` and getting the following error.

```
Warning: com.mtramin.rxfingerprint.FingerprintObservable$$Lambda$1: can't find referenced class java.lang.invoke.LambdaForm$Hidden

```

Works fine on adding the following rule to proguard : 

```
#rxFingerprint
-dontwarn com.mtramin.rxfingerprint.**
```

Mauin, 2017-01-30:
Thanks for the report @ragdroid. I'll investigate this one!

Mauin, 2017-02-02:
This will go out with the next release. I'll first release the 2.x variant and then work on backporting all the changes to 1.x

closed 2017-02-02 by Mauin

Mauin, 2017-02-05:
I just released v2.0.2 which should fix this issue. I'll start working on backporting the changes to the `1.x` branch but I can't give an ETA on the next `1.x` release.

Mauin, 2017-02-08:
Just released v1.2.3 which should not need that additional ProGuard rule anymore. Let me know if this causes any more issues!

PR #29 (2017-02-02): Remove Retrolambda usages from library
Retrolambda was just used once in the library. 

As seen in #27 this would require users who don't have Retrolambda set up to add additional ProGuard rules.

- resolves #27