Mauin/RxFingerprint #30: Consider using FingerprintManager instead of FingerprintManagerCompat

Mauin, 2017-02-02:
Some devices that have a fingerprint sensor are not specifying the system feature `FEATURE_FINGERPRINT` correctly. The support library is explicitly checking for the system feature. 

This leads devices which actually have a fingerprint sensor to report `false` for `RxFingerprint.isHardwareDetected` check.

Using the `FingerprintManager` directly and not doing this check should allow the offending devices to be used with the library. However it is not guaranteed that all devices which do _not_ specify the `FEATURE_FINGERPRINT` feature even implement the Fingerprint API classes correctly.

AOSP issue: https://code.google.com/p/android/issues/detail?id=231939

mikeKickserv, 2017-02-02:
I'm running into this. When is the ETA for a fix for this?

Mauin, 2017-02-02:
I'm working on it right now. I'd love to get some other fixes into the next release. Should be released on the weekend. 

Do you need a 2.x or 1.x variant?

mikeKickserv, 2017-02-02:
I'm currently using 2.0.1 so 2.x. Thanks!

closed 2017-02-05 by Mauin

Mauin, 2017-02-05:
Just released v2.0.2 which removes `FingerprintManagerCompat` usages. That should fix the issues you're seeing. Feel free to reopen this if you still run into issues with those devices.

PR #31 (2017-02-05): Use FingerprintManager instead of FingerprintManagerCompat
Removes all AppCompat usages from RxFingerprint. All usages of FingerprintManagerCompat are replaced by FingerprintManager directly to work around the wrongly defined system features in some devices.

Adds checking the SDK version manually which was previously done in AppCompat. 

Resolves #30