Mauin/RxFingerprint #42: java.lang.NullPointerException: Attempt to invoke interface method 'int java.util.List.size()' on a null object reference

Farasy, 2017-02-20:
Thanks for your great work.
I'm using your lib for a while and run some tests but currently getting the following error when trying to init RxFingerprint.decrypt()

`                                                              java.lang.NullPointerException: Attempt to invoke interface method 'int java.util.List.size()' on a null object reference
                                                                  at android.os.Parcel.readException(Parcel.java:1626)
                                                                  at android.os.Parcel.readException(Parcel.java:1573)
                                                                  at android.hardware.fingerprint.IFingerprintService$Stub$Proxy.authenticate(IFingerprintService.java:300)
                                                                  at android.hardware.fingerprint.FingerprintManager.authenticate(FingerprintManager.java:601)
                                                                  at android.hardware.fingerprint.FingerprintManager.authenticate(FingerprintManager.java:574)
                                                                  at android.hardware.fingerprint.FingerprintManager.authenticate(FingerprintManager.java:552)
                                                                  at com.mtramin.rxfingerprint.FingerprintObservable.subscribe(FingerprintObservable.java:81)
                                                                  at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
                                                                  at io.reactivex.Observable.subscribe(Observable.java:10179)
                                                                  at io.reactivex.Observable.subscribe(Observable.java:10165)
                                                                  at io.reactivex.Observable.subscribe(Observable.java:10096)
                                                                  at gui.Login.initFingerprints(Login.java:179)
                                                                  at gui.Login.onResume(Login.java:1179)
                                                                  at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1287)
                                                                  at android.app.Activity.performResume(Activity.java:7015)
                                                                  at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4210)
                                                                  at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4323)
                                                                  at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1865)
                                                                  at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                  at android.os.Looper.loop(Looper.java:148)
                                                                  at android.app.ActivityThread.main(ActivityThread.java:7325)
                                                                  at java.lang.reflect.Method.invoke(Native Method)
                                                                  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)`

Source

`                Disposable disposable = RxFingerprint.decrypt(this, "k", AppSettings.getFingerprintHash(getAppContext())). //line 179
subscribe(fingerprintDecryptionResult -> {
                    if(fingerprintDecryptionResult != null){
                        switch(fingerprintDecryptionResult.getResult()){
                            case FAILED:
                                setStatusText("Fingerprint not recognized, try again!");
                                break;
                            case HELP:
                                setStatusText(fingerprintDecryptionResult.getMessage());
                                break;
                            case AUTHENTICATED:
                                setStatusText("decrypted:\n" + fingerprintDecryptionResult.getDecrypted());
                                break;
                        }
                    }
                }, throwable -> {
                    //noinspection StatementWithEmptyBody
                    if(RxFingerprint.keyInvalidated(throwable)){
                        // The keys you wanted to use are invalidated because the user has turned off his
                        // secure lock screen or changed the fingerprints stored on the device
                        // You have to re-encrypt the data to access it
                    }
                    Log.e("ERROR", "decrypt", throwable);
                });`

Any idea where it comes from?

Farasy, 2017-02-21:
My fault, the above message is just a result of a interrupt in listening to the fingerprint.

closed 2017-02-21 by Farasy

Mauin, 2017-02-21:
@Farasy Glad you figured it out. How did you reproduce this issue with the interrupt?

Farasy, 2017-02-23:
I got an exception thrown from a service. My tired eyes haven't seen it some lines above the stacktrace.. Assumed that the problem comes from RxFingerprint but it didn't.

kumawatmanish, 2018-09-07:
@Farasy How did you reproduce this issue? Actually, I got crash logs, but unable to reproduce this issue.

Farasy, 2018-09-26:
> @Farasy How did you reproduce this issue? Actually, I got crash logs, but unable to reproduce this issue.

Missed your question, sorry.. but i guess i can't help you. I did some tests with a service that ended in above message. However, don't know what i did exactly (anymore ) oO

hvge, 2019-08-21:
Hi guys, I discovered this ticket while was working on a similar problem. Check out my last comment in wultra/powerauth-mobile-sdk#202, if this is still an issue, of course :)