Mauin/RxFingerprint #47: Callback is not called second time on error

loentar, 2017-04-11:
When you start example and use unregistered finger to cause an error, the callback(Consumer) is not called second time. But it works fine every time you use registered finger.

I use `dispose()` inside callback to deinit sensor. After that second call works.

Mauin, 2017-04-14:
@loentar I just tried to reproduce the issue, but I'm not able to do so.

When trying to authenticate and using an unregistered finger on the sensor you will receive an `onNext` event with the `FingerprintAuthenticationResult.FAILED` reason. Even if you use an unregistered finger multiple times you will receive this event every time. 

However, after a couple of wrong tries the fingerprint sensor will lock down, cancel the authentication action and not allow any more fingerprint authentication for some time.

Once you call `dispose()` the fingerprint authentication will immediately be cancelled. So if you call `dispose()` in the `onNext` callback, you will not be able to receive a 2nd event. Can you share the code you're using with RxFingerprint? Then I can give more input.

loentar, 2017-04-14:
I use example provided by library.

Please see this screencast for details: https://drive.google.com/file/d/0BwvdX1pBENi1Z0ZPTWtHNGl6WE0/view

Mauin, 2017-04-17:
Looking at the screencast you're trying to authenticate too often with the finger that is not registered. That will cause the case I've described above: The fingerprint sensor locks down for a short amount of time. 

When reproducing the screencast, I can see the following in logcat:
```
E/ERROR: encrypt
    com.mtramin.rxfingerprint.data.FingerprintAuthenticationException: Too many attempts. Try again later.
```

This error is coming directly from the `FingerprintManager` of the Android SDK and RxFingerprint will forward this error in the `onError` of your subscription. The example application handles this error only by logging it to logcat and doesn't update the status line accordingly.

Mauin, 2017-04-17:
#49 will now show the errors in the sample app accordingly and update the status line.

PR #49 (2017-04-17): show onError messages in sample app
Resolves #47 

Shows the error throwable message in the status line of the sample application.
closed 2017-04-17 by Mauin