Mauin/RxFingerprint #52: Unexposed Internal Crash

ScottPierce, 2017-05-18:
We are seeing a crash in the latest version of the library that isn't being exposed for us to handle. It seems to be related to changing the fingerprint in Android settings.

Because the crash isn't exposed to our code at all, we have no way of catching the exception and handling it in some way. It forces a crash on our application, and there isn't anything we can do.

```java
Fatal Exception: java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
Caused by java.lang.reflect.InvocationTargetException
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
Caused by javax.crypto.IllegalBlockSizeException
       at android.security.keystore.AndroidKeyStoreCipherSpiBase.engineDoFinal(AndroidKeyStoreCipherSpiBase.java:519)
       at javax.crypto.Cipher.doFinal(Cipher.java:2056)
       at com.mtramin.rxfingerprint.AesEncryptionObservable.onAuthenticationSucceeded(AesEncryptionObservable.java:95)
       at com.mtramin.rxfingerprint.FingerprintObservable$2.onAuthenticationSucceeded(FingerprintObservable.java:104)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.sendAuthenticatedSucceeded(FingerprintManager.java:876)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:824)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6290)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
Caused by android.security.KeyStoreException: Key user not authenticated
       at android.security.KeyStore.getKeyStoreException(KeyStore.java:666)
       at android.security.keystore.KeyStoreCryptoOperationChunkedStreamer.update(KeyStoreCryptoOperationChunkedStreamer.java:132)
       at android.security.keystore.KeyStoreCryptoOperationChunkedStreamer.doFinal(KeyStoreCryptoOperationChunkedStreamer.java:217)
       at android.security.keystore.AndroidKeyStoreCipherSpiBase.engineDoFinal(AndroidKeyStoreCipherSpiBase.java:506)
       at javax.crypto.Cipher.doFinal(Cipher.java:2056)
       at com.mtramin.rxfingerprint.AesEncryptionObservable.onAuthenticationSucceeded(AesEncryptionObservable.java:95)
       at com.mtramin.rxfingerprint.FingerprintObservable$2.onAuthenticationSucceeded(FingerprintObservable.java:104)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.sendAuthenticatedSucceeded(FingerprintManager.java:876)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:824)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6290)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
```

Mauin, 2017-05-18:
Oh noes! That definitely shouldn't happen. All errors should be returned as an `onError` callback. But seems like this one is not. 
Is this happening across devices or does it look specific to one device?

I'll take a look at this! Thanks for reporting!

ScottPierce, 2017-05-18:
We are seeing this across multiple devices on multiple versions of Android.

Mauin, 2017-05-21:
This leaves me a bit puzzled. Do you have any logs that this might've also happened with previous versions but you received the Errors in `onError()`? Just as a sanity check: You're handling the `onError()` callback of the RxFingerprint subscription, right?

I tried reproducing this with the sample and manually throwing the errors, but as the `onAuthenticationSucceeded` callback catches all errors thrown during encryption and forwards them to `onError()` I'm not quite sure how to reproduce this. 

I'm also seeing this thread on SO claiming that Samsung, Acer and OnePlus devices seem to have this at times: https://stackoverflow.com/questions/36043912/error-after-fingerprint-touched-on-samsung-phones-android-security-keystoreexce 
Does this match what you're seeing?

ScottPierce, 2017-05-21:
Yes I'm handling the `onError()` callback. I'm seeing this happen with Samsung and Nexus 5X devices. I don't have many other devices to test it with.

I saw that SO error also. To reproduce this, it's super edge case, and seems more like a bug that I'd like to catch, than to change my KeySpec to handle it.

Using the StackTrace, we should be able to see where the error is coming from, and ensure that it gets routed to `onError()` right?

Mauin, 2017-05-22:
I'll see if I can get my hands on any of those devices to try to reproduce this issue.

From the stacktrace, the Exception gets thrown here: https://github.com/Mauin/RxFingerprint/blob/master/rxfingerprint/src/main/java/com/mtramin/rxfingerprint/AesEncryptionObservable.java#L95

But that code has a `try/catch` block around it that should forward all Exceptions to `onError()` to give you a way to handle it. That's what is puzzling me.

Mauin, 2017-06-06:
@ScottPierce Are you able to reproduce this issue with any devices you have on hand? If yes I'd be grateful if you can debug what's going on in RxFingerprint. 
I'm unable to reproduce this with any of my devices. As mentioned above I expect the `try/catch` block to catch all Exceptions and forward them to the caller of the library in `onError()`.

Mauin, 2017-12-12:
This should be resolved with the fixes in the latest releases. Feel free to reopen this one if the issue sticks around.

closed 2017-12-12 by Mauin