Mauin/RxFingerprint #56: Invalidate keys after enrolling an additional fingerprint

mato666, 2017-06-14:
After enroling an additional fingerprint, there should be option to invalidate stored keys. In Android 6 this is default behaviour, but on Android 7 you must set this in KeyGenParameterSpec builder (setInvalidatedByBiometricEnrollment).

Please see following post: https://benlikestoco.de/notable-fingerprint-api-changes-in-android-7-cc3ec99a11c4

You could maybe open option for defining some attributes, when building KeyGenParameterSpec. Anyway I thing setInvalidatedByBiometricEnrollment(true) should be set by default.

Thanks.

Mauin, 2017-06-14:
Great suggestion! I agree with the default suggestion.
Do you want to create a PR that sets `setInvalidatedByBiometricEnrollment(true)` by default on Android 7?

As for setting this via each call manually in the `RxFingerprint` API I'm a bit concerned with the amount of arguments the `encrypt`/`decrypt` calls already take and would like to find a way to simplify how to call them while still giving the users of the library the option to set things like this.

I might think about giving `RxFingerprint` a builder in which users can set their preferences and only pass the required arguments to `encrypt`/`decrypt`.

mato666, 2017-06-14:
I have never done pull request before :) so could you please add mentioned line in code.
Idea about adding a builder to set additional preferences is great and is also future proof.

ahmaadyunus, 2017-10-23:
I have same problem while i running apps on emulator.enrolled finger print not read. Solution: Delete finger print exist and add again.

Mauin, 2017-11-27:
Closed via #67

closed 2017-11-27 by Mauin