Mauin/RxFingerprint #57: Expose errMsgId to subscribers

joshfriend, 2017-06-15:
If the fingerprint authentication fails with an error like [`FINGERPRINT_ERROR_LOCKOUT`](https://developer.android.com/reference/android/hardware/fingerprint/FingerprintManager.html#FINGERPRINT_ERROR_LOCKOUT), the UI should display an error for a longer period of time than if the error is `FINGERPRINT_ACQUIRED_TOO_FAST` for example. There is currently no way to know what the specific error condition of the reader is. The error id should be passed along inside `FingerprintAuthenticationException`

Mauin, 2017-06-15:
I can think about passing the ID of the error message along. 

Generally all errors that cause the Fingerprint sensor to "lock down" (`FINGERPRINT_ERROR_LOCKOUT` is one example of this)  will be passed in `.onError()`. Recoverable Errors such as moving the finger too fast will be delivered in `.onNext()`. This distinction between `.onNext` and `.onError` already gives a hint at the severity of the issue.

joshfriend, 2017-06-15:
Thanks for pointing that out! It does help a lot, but if the subscriber is disposed on rotation, `cancellationSignal.cancel()` causes `FINGERPRINT_ERROR_CANCELED` to be sent to the `onAuthenticationError` callback. It would be nice to be able to distinguish this from an error condition that should be shown to the user.