Mauin/RxFingerprint #58: FingerprintAuthenticationException

eGorets, 2017-06-20:
I just got error log in Fabric. I don't have more information about this. Please advice me how to fix this if you can. Thanks
```
Fatal Exception: com.mtramin.rxfingerprint.data.FingerprintAuthenticationException
Vingerafdrukbewerking geannuleerd.
 Raw Text
com.mtramin.rxfingerprint.FingerprintObservable$2.onAuthenticationError (FingerprintObservable.java:98)
android.hardware.fingerprint.FingerprintManager$MyHandler.sendErrorResult (FingerprintManager.java:859)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:776)
```

Mauin, 2017-06-20:
Which version of RxFingerprint are you using?

eGorets, 2017-06-20:
`2.0.3`, I see you released new one. I'm updating to it right now. Also crash come from ONE PLUS A3003 phone and I can't test it

Mauin, 2017-06-20:
Are you handling the `onError(Throwable)` callback in the Observer that you use to subscribe to RxFingerprint?

eGorets, 2017-06-20:
Nope. I thought it had to be a pretty simple use-case.

my code is 

```

  if (RxFingerprint.isAvailable(getActivity())) {
            RxFingerprint.authenticate(getActivity())
                    .compose(this.<FingerprintAuthenticationResult>bindToLifecycle())
                    .subscribe(new Consumer<FingerprintAuthenticationResult>() {
                        @Override
                        public void accept(FingerprintAuthenticationResult fingerprintAuthenticationResult) throws Exception {
                            switch (fingerprintAuthenticationResult.getResult()) {
                                case FAILED:
                                    break;
                                case HELP:
                                    break;
                                case AUTHENTICATED:
                                    passwordAccepted();
                                    break;
                            }
                        }
                    });
        }
```

Mauin, 2017-06-20:
You should always handle the `onError` values. In this case you're not. Now every time you dispose the RxFingerprint subscription the fingerprint sensor will report "Operation Cancelled" which will cause an `onError`. Because they're not handled at all they can crash the application.

eGorets, 2017-06-20:
Got it. Thanks. My thought was that there was only 3 possible states of `FingerprintAuthenticationResult`. but `onError` worked only for more diff cases like megring with network calls or cryptography

closed 2017-06-20 by eGorets

Mauin, 2017-06-20:
`onError` will get called for all events that cause the Fingerprint sensor to stop listening for fingerprint events. This can be because RxFingerprint tells it to stop, because an unknown fingerprint tried to authenticate too many times, because the sensor is broken, or other issues.

betomaluje, 2017-12-15:
I was having this issue with the latest version 'com.mtramin:rxfingerprint:2.2.1'. After a time trying everything, I just restarted the phone (reboot) and....everything worked as normal :/

Huawei Honour 8 hahaha