Mauin/RxFingerprint #63: Authentication Required

axelwerner, 2017-07-21:
Hello,

I want to know if its possible to set setUserAuthenticationRequired and setUserAuthenticationValidityDurationSeconds from the dependence. From that what I can see, I don't have the possibility to set this, is that right? If yes, how??

Thanks in advance

Mauin, 2017-07-21:
`setUserAuthenticationRequired` is set by default for all Ciphers used for encryption/decryption in RxFingerprint (both AES and RSA).
It is indeed not possible to provide a value for `setUserAuthenticationValidityDurationSeconds` at the moment. 
However I'm currently thinking about a refactor to enhance the public API of RxFingerprint to open up more customization to the users of RxFingerprint. I'll take `setUserAuthenticationValidityDurationSeconds` into account for that as well.

axelwerner, 2017-07-21:
Thanks for your quickly answer.
It would be nice to have more customization possibilities. 
Nice API 👍 

Thank you

iprasla, 2017-12-04:
I would be interested in having the authentication validation duration set. In my case, once I authenticate the user, I present the user account selection (user credentials saved) to be able to retrieve encrypted credentials. With current limitations, I need to present user twice with fingerprint sensor. First for prior to account selection and second time when retrieving the credentials.

Mauin, 2017-12-05:
This would be a great addition to RxFingerprint. To not further add more overloaded methods to `RxFingerprint` for encryption/decryption I think this would be a great configuration to have in the Builder pattern I started working on in #72. 
Then the builder could take an additional method: `builder. setUserAuthenticationValidityDurationSeconds(duration)` to set this up and use this value for all encryption/decryption operations.