Mauin/RxFingerprint #64: OnePlius 3 capacitive button dismiss the app

stepansanda, 2017-08-01:
I have implemented fingerprint login to my app with RxFingerprint. Now I found that the fingerprint sensor on OnePlus 3 is placed on the home button that is, opposite to other phones, touch button (not clicky one). So when the user tries to login with his fingerprint, he touch the home button, the app goes to the background instead of read fingerprint and login into the app.

I'm curious, if I missed some step that is handling this situation, or if there have to be some workaround to fix this issue? 

As I understand in normal implementation according to Google Sample it works correctly (it is working in other apps on OP3) - listening for fingerprint should disable home button click in that case.

Mauin, 2017-08-03:
Just so that I understand the issue correctly: Other apps with Fingerprint Authentication disable the home button press while the fingerprint sensor is active?

It sounds like that should be a system feature and the regular behavior. All `RxFingerprint` does is tell the `FingerprintManager` to start the authentication flow.

stepansanda, 2017-08-03:
@Mauin This is definitely the system feature and regular behavior. As I wrote in my update above, I found that one of my colleague has OP3 and he has no fingerprint issue with my app. But one of our customer is reporting this behavior. So maybe it is only some device settings issue. Right now I'm still investigating where can be the problem

closed 2017-12-12 by Mauin