Mauin/RxFingerprint #65: Add sign and verify functions for RSA

Zookey, 2017-08-28:
Hi,

After successful fingerprint authentication I need to sign data with RSA and then send it to the server for verifying.

I think that those functions could be interesting to more users. Please consider adding them.

Mauin, 2017-08-28:
`RxFingerprint` provides functionalities around the Fingerprint API. It does handle RSA for encrypting and decrypting values via the Fingerprint APIs `CryptoObject`. Is that what you need for signing your data?

Zookey, 2017-08-28:
Yeah, just instead of encrypting/decrypting I need to sign data and then later verify it on the server.

Mauin, 2018-01-21:
Coming back to this: How would you like an API for this to look like? What are the inputs/outputs?

Mauin, 2018-03-16:
Closing this for inactivity. Feel free to re-open this issue with more info.

closed 2018-03-16 by Mauin