Mauin/RxFingerprint #66: IllegalBlockSizeException on some devices

xdgimf, 2017-09-12:
We have implemented fingerprint authentication in our app. But for some reason few devices throws IllegalBlockSizeException when using the encrypt api. 

```
RxFingerprint.encrypt(getActivity(), fingerprintToken)
                .compose(bindToLifecycle())
                .subscribe(result -> {
                    switch (result.getResult()) {
                        case AUTHENTICATED:
                            ...
                            break;
                    }
                }, error -> {
                    // Few devices throws IllegalBlockSizeException when user enters fingerprint
                });
```

List of Devices from Fabric
Fingerprint Error
Flash Plus 2
Infinix_X521
HOT 4
Lenovo K5 Note
Flare X2
Infinix NOTE 3 Pro

```
Non-fatal Exception: javax.crypto.IllegalBlockSizeException
       at android.security.keystore.AndroidKeyStoreCipherSpiBase.engineDoFinal(AndroidKeyStoreCipherSpiBase.java:486)
       at javax.crypto.Cipher.doFinal(Cipher.java:1502)
       at com.mtramin.rxfingerprint.FingerprintEncryptionObservable.onAuthenticationSucceeded(Unknown Source:112)
       at com.mtramin.rxfingerprint.FingerprintObservable$2.onAuthenticationSucceeded(Unknown Source:114)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.sendAuthenticatedSucceeded(FingerprintManager.java:882)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:834)
       at android.os.Handler.dispatchMessage(Handler.java:111)
       at android.os.Looper.loop(Looper.java:207)
       at android.app.ActivityThread.main(ActivityThread.java:5765)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:789)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:679)
Caused by android.security.KeyStoreException: Key user not authenticated
       at android.security.KeyStore.getKeyStoreException(KeyStore.java:650)
       at android.security.keystore.KeyStoreCryptoOperationChunkedStreamer.update(KeyStoreCryptoOperationChunkedStreamer.java:132)
       at android.security.keystore.KeyStoreCryptoOperationChunkedStreamer.doFinal(KeyStoreCryptoOperationChunkedStreamer.java:217)
       at android.security.keystore.AndroidKeyStoreCipherSpiBase.engineDoFinal(AndroidKeyStoreCipherSpiBase.java:473)
       at javax.crypto.Cipher.doFinal(Cipher.java:1502)
       at com.mtramin.rxfingerprint.FingerprintEncryptionObservable.onAuthenticationSucceeded(Unknown Source:112)
       at com.mtramin.rxfingerprint.FingerprintObservable$2.onAuthenticationSucceeded(Unknown Source:114)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.sendAuthenticatedSucceeded(FingerprintManager.java:882)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:834)
       at android.os.Handler.dispatchMessage(Handler.java:111)
       at android.os.Looper.loop(Looper.java:207)
       at android.app.ActivityThread.main(ActivityThread.java:5765)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:789)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:679)
```

mato666, 2017-11-27:
I think you get this error on devices that run Android 7 and higher, when you add additional fingerprint. After that when you try to encrypt/decrypt some text, you get this error (related to #56).
Workaround until issue is resolved (pull request already done #67):

```
private static final String ANDROID_KEY_STORE = "AndroidKeyStore";
private static final String FINGERPRINT_KEY_NAME = "dummy";

...

boolean keyInvalidatedWorkaround = throwable instanceof IllegalBlockSizeException;
if (keyInvalidatedWorkaround) {
    removeKey(FINGERPRINT_KEY_NAME);
}

private static boolean keyExists(String keyName) throws KeyStoreException, CertificateException, NoSuchAlgorithmException, IOException {
    KeyStore keyStore = KeyStore.getInstance(ANDROID_KEY_STORE);
    keyStore.load(null);
    Enumeration<String> aliases = keyStore.aliases();

    while (aliases.hasMoreElements()) {
        if (keyName.equals(aliases.nextElement())) {
            return true;
        }
    }

    return false;
}

private static void removeKey(String keyName) throws KeyStoreException, CertificateException, NoSuchAlgorithmException, IOException {
    if (keyExists(keyName)) {
        KeyStore keyStore = KeyStore.getInstance(ANDROID_KEY_STORE);
        keyStore.load(null);
        keyStore.deleteEntry(keyName);
    }
}
```

mato666, 2017-11-28:
Found possible issue:  Android Oreo doesn't throw KeyPermanentlyInvalidatedException https://issuetracker.google.com/issues/65578763
https://github.com/googlesamples/android-FingerprintDialog/issues/21
It seems that only Android 8 phones have this problem. So workaround could be:

`boolean keyInvalidatedWorkaround = Build.VERSION.SDK_INT >= Build.VERSION_CODES.O && throwable instanceof IllegalBlockSizeException;`

Mauin, 2017-11-28:
I think it makes sense putting that workaround in the library to "emulate" throwing a `KeyPermanentlyInvalidatedException` in those cases. That issue in the bugtracker certainly looks convincing.

@mato666 Do you want to work on a PR for this? Otherwise I'll take a look at it and will try to repro this.

mato666, 2017-11-28:
@Mauin I would like to.

Mauin, 2017-11-28:
@mato666 Awesome! Go for it!

mato666, 2017-11-29:
Made PR: #71.

Mauin, 2017-12-12:
Closing this as the PR is merged and the change was released in v2.2.1