Mauin/RxFingerprint #74: Exception when restoring app

Flo354, 2017-12-13:
Hi,

I am facing an issue when an user restore my app with TitaniumBackup or synchronize it from another phone (example: samsung). Here is the exception: 

```
java.security.InvalidKeyException: Key for algorithm null not suitable for symmetric enryption.
    at org.spongycastle.jcajce.provider.symmetric.util.BaseBlockCipher.engineInit(BaseBlockCipher.java:496)
    at javax.crypto.Cipher.tryTransformWithProvider(Cipher.java:2973)
    at javax.crypto.Cipher.tryCombinations(Cipher.java:2884)
    at javax.crypto.Cipher$SpiAndProviderUpdater.updateAndGetSpiAndProvider(Cipher.java:2789)
    at javax.crypto.Cipher.chooseProvider(Cipher.java:956)
    at javax.crypto.Cipher.init(Cipher.java:1329)
    at javax.crypto.Cipher.init(Cipher.java:1267)
    at com.mtramin.rxfingerprint.AesCipherProvider.getCipherForDecryption(AesCipherProvider.java:87)
    at com.mtramin.rxfingerprint.AesDecryptionObservable.initCryptoObject(AesDecryptionObservable.java:83)
    at com.mtramin.rxfingerprint.FingerprintObservable.subscribe(FingerprintObservable.java:70)
    ...
```

The error comes from this portion of code in file *AesCipherProvider.java*.

```java
	/**
	 * @param iv initialization vector used during encryption
	 * @return Initialized cipher for decryption operations in RxFingerprint
	 */
	Cipher getCipherForDecryption(byte[] iv) throws CertificateException, NoSuchAlgorithmException, IOException, InvalidKeyException, UnrecoverableKeyException, KeyStoreException, InvalidAlgorithmParameterException, NoSuchPaddingException {
		Cipher cipher = createCipher();
		SecretKey key = getKey(keyName);
		cipher.init(Cipher.DECRYPT_MODE, key, new IvParameterSpec(iv));
		return cipher;
	}
```

Do you think it would be ok to detect this behavior and instead throw an InvalidatedKeyException with the removal of the key?

Thanks,

Mauin, 2017-12-17:
This seems like a really edge case scenario. Due to the backup/restore of the application data the app seems to believe that the encrypted data is still valid however the Android KeyStore does not contain that key, leading to this exception to be thrown. This sounds like valid behavior to me. Similar to what would happen if you called `decrypt()` before ever encrypting anything.

Mauin, 2018-03-16:
Closing for inactivity. This seems really edge-case. Feel free to re-open this with more info if needed.

closed 2018-03-16 by Mauin