Mauin/RxFingerprint #75: Memory Leak

siddharthuohyd, 2017-12-18:
Getting leak of 4.7KB using this library.Added the screenshot also for better clarity.This is the snippet that I have used in my project which gets the leak which is determined by the Leak Canary tool.

 private void fingerprintAuthentication() {
        if (RxFingerprint.isAvailable(getAppContextFromView())) {
            mDisposable = RxFingerprint.authenticate(getAppContextFromView())
                    .subscribeOn(Schedulers.io())
                    .subscribe(fingerprintAuthenticationResult -> {
                        switch (fingerprintAuthenticationResult.getResult()) {
                            case FAILED:
                                Toast.makeText(getAppContextFromView(),
                                        R.string.error_finger_print_not_valid,
                                        Toast.LENGTH_SHORT).show();
                                if (mFingerPrintTouchCallBack != null) {
                                    mFingerPrintTouchCallBack.onAuthenticationFailed();
                                }

                                break;
                            case HELP:
                                Toast.makeText(getAppContextFromView(),
                                        fingerprintAuthenticationResult.getMessage(),
                                        Toast.LENGTH_SHORT).show();
                                break;
                            case AUTHENTICATED:
                                if (mFingerPrintTouchCallBack != null) {
                                    mFingerPrintTouchCallBack.onAuthenticationSuccess();
                                    mConfirmPinPresenter.pinUpdated();
                                }
                                break;
                        }
                    }, throwable -> {
                        if (throwable instanceof FingerprintAuthenticationException) {
                            if (throwable.getMessage() != null && throwable.getMessage()
                                    .contains(
                                            TOO_MANY_ATTEMPT_STR)) {
                                Toast.makeText(getAppContextFromView(),
                                        throwable.getMessage(),
                                        Toast.LENGTH_SHORT).show();
                                mConfirmPinPresenter.logout();
                            }
                        }
                    });
        }
    }

[image]

Mauin, 2017-12-22:
Thanks for reporting this leak. I'll take a look at it and see if there is anything we can improve in RxFingerprint or if the `FingerprintManager` internally is causing this leak.

kotomisak, 2019-02-11:
I had similar issue and it seems to be FingerprintManager issue. The workaround is described here: https://stackoverflow.com/questions/53481621/memory-leak-but-how-can-i-pass-a-different-context-than-the-one-of-the-activity