Mauin/RxFingerprint #76: RSA Decryption throwing exception?

iprasla, 2017-12-18:
My production code is crashing and the crash log from the play store ANRs & Crashes give me following log. What could be causing this?

`io.reactivex.exceptions.CompositeException: 
  at io.reactivex.internal.observers.LambdaObserver.onError (LambdaObserver.java:77)
  at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError (ObservableSubscribeOn.java:63)
  at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError (ObservableCreate.java:85)
  at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.onError (ObservableCreate.java:73)
  at com.mtramin.rxfingerprint.RsaDecryptionObservable.initCryptoObject (RsaDecryptionObservable.java:79)
  at com.mtramin.rxfingerprint.FingerprintObservable.subscribe (FingerprintObservable.java:70)
  at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual (ObservableCreate.java:40)
  at io.reactivex.Observable.subscribe (Observable.java:10838)
  at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run (ObservableSubscribeOn.java:96)
  at io.reactivex.Scheduler$DisposeTask.run (Scheduler.java:452)
  at io.reactivex.internal.schedulers.ScheduledRunnable.run (ScheduledRunnable.java:61)
  at io.reactivex.internal.schedulers.ScheduledRunnable.call (ScheduledRunnable.java:52)
  at java.util.concurrent.FutureTask.run (FutureTask.java:266)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run (ScheduledThreadPoolExecutor.java:301)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1162)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:636)
  at java.lang.Thread.run (Thread.java:764)`

Mauin, 2017-12-22:
Thanks for reporting this issue. Do you have more info in the stack trace? The current one doesn't actually show which exception is getting thrown.

iprasla, 2017-12-26:
Unfortunately that is all I could get from the crash. This is the crash reported in production and that is all it was posted in the ANRs & crashes in Playstore.

iprasla, 2018-01-02:
I found more details on the above error:

```
01-02 15:47:46.587 7261-7301/? E/ERROR: decrypt
                                        java.security.InvalidKeyException: RSA private or public key is null
                                            at com.android.org.conscrypt.OpenSSLCipherRSA.engineInitInternal(OpenSSLCipherRSA.java:205)
                                            at com.android.org.conscrypt.OpenSSLCipherRSA.engineInit(OpenSSLCipherRSA.java:221)
                                            at javax.crypto.Cipher.tryTransformWithProvider(Cipher.java:2668)
                                            at javax.crypto.Cipher.tryCombinations(Cipher.java:2575)
                                            at javax.crypto.Cipher$SpiAndProviderUpdater.updateAndGetSpiAndProvider(Cipher.java:2480)
```

PingForward, 2018-02-01:
It seems I have something similar
```
Fatal Exception: io.reactivex.exceptions.CompositeException: 2 exceptions occurred. 
       at io.reactivex.internal.observers.LambdaObserver.onError(LambdaObserver.java:77)
       at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:85)
       at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.onError(ObservableCreate.java:73)
       at com.mtramin.rxfingerprint.RsaDecryptionObservable.onAuthenticationSucceeded(RsaDecryptionObservable.java:95)
       at com.mtramin.rxfingerprint.FingerprintObservable$2.onAuthenticationSucceeded(FingerprintObservable.java:104)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.sendAuthenticatedSucceeded(FingerprintManager.java:1217)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:1148)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6776)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)
```

iprasla, 2018-02-01:
@PingForward Are you using dialog fragment? My issue was resolved by checking if the fragment was added before or not.

Mauin, 2018-02-01:
@PingForward The stacktrace you shared seems like a different issue. It looks like RxFingerprint is not able to correctly decrypt the value you passed in. Was the value previously encrypted by RxFingerprint with RSA?

PingForward, 2018-02-02:
@iprasla No, this is regular fragment (MvpAppCompatFragment from com.arello-mobile:moxy lib)
@Mauin Yes, the encrypted string is correct
I used decript function in onStart() of fragment
```
fun checkIfCanUseFingerPrint(context: Context) {
        if (isFingerprintAvailable(context)) {
            val encryptedString = getEncryptedPassword()
            if (encryptedString.isNotEmpty()) {
                startListenFingerPrint(context, encryptedString)
            }
        } 
    }
```
**startListenFingerPrint**
 ```
disposableDecript = RxFingerprint.decrypt(EncryptionMethod.RSA, context, key, encriptedString)
                .subscribe({ decryptionResult ->
                 ***
                }, { throwable ->
                    if (RxFingerprint.keyInvalidated(throwable)) {
                      ***
                    }
                    **Log.d("RxFingerprint", throwable.message)** // error here, but it's not NPE
                })
    }
```

I dispose disposableDecript in onDestroy
maybe I should move it in to onStop, hmm

Mauin, 2018-02-02:
Keeping the same lifecycle between subscribing and disposing would be a good idea to try I think `onStart()` -> `onStop()`

Mauin, 2018-03-16:
I'm closing this for inactivity. If you see further issues feel free to re-open this issue or create a new one.

closed 2018-03-16 by Mauin