Mauin/RxFingerprint #77: Question - Some explanation on RSA implementation

kuno, 2017-12-27:
Hi: 
    I am new to cryptography and thanks for the great library.

   I have two questions on the RSA implementation of this lib.
*  I saw that the RSA Cipher was created with ECB operation mode https://github.com/Mauin/RxFingerprint/blob/master/rxfingerprint/src/main/java/com/mtramin/rxfingerprint/RsaCipherProvider.java#L55

    I heard that ECB is not safe and should not be been using in production.

    Can you explain a bit why you choosing ECB?

 * How can I set the size of RSA private key?

kuno, 2017-12-28:
Another question is that Is that safe to use AEA on devices that before API level 23 ?

I also heard that Android keystore only support AES since API 23.

Mauin, 2017-12-30:
Hey, thanks for creating this issue. As for the 2nd part: The Fingerprint APIs were only added to the Android SDK in Android API 23. So it is not possible to use AES with this library with API levels before 23.
I'll do some more reading about RSA Ciphers with ECB Block mode.