Mauin/RxFingerprint #82: Offer char[] Option For Data to encrypt() and decrypt()

commonsguy, 2018-01-20:
The `encrypt()` method takes the cleartext data as a `String`; the `decrypt()` method provides the decrypted cleartext data as a `String`. The problem in Java is that strings are immutable, which means we cannot forcibly clear their contents.

In some cases, the cleartext data is the actual data you are acting upon (e.g., you are encrypting/decrypting a JSON string, where the JSON represents your app's actual data). In that case, your cleartext data will be around as long as your app needs it anyway.

In some cases, though, the cleartext data is something that you would like to get rid of quickly. For example, perhaps the cleartext data is a generated passphrase to use with SQLCipher for Android or some other non-keystore encrypted container. In this case, it would be far preferable for the cleartext data to be `char[]` instead of `String`, as we can change the bytes of the `char[]` to erase the cleartext data once we are done with it.

For example, you could have:

- `encrypt()` variants that take a `char[]` for `toEncrypt`
- have `FingerprintDecryptionResult` hold a `char[]` for the decrypted data, with a getter method to retrieve it, and modifying `getDecrypted()` to convert the `char[]` into a `String`

From what I can tell, those changes should keep a compatible API with what you have, while still offering `char[]` support for those who need it.

Anyway, it's just an idea. Thanks for writing this library!

Mauin, 2018-01-21:
Sounds like a great idea! As you already described it would result in more `encrypt()` methods that take in the `char[]` and also offering the `char[]` in the result object.

While this would result in quite a few more methods in the `RxFingerprint` class it seems worthwhile. The refactor to move RxFingerprint to use the Builder pattern in #72 should also reduce the amount of `encrypt()` methods by a bit.

Do you want to create a PR to add support for `char[]`? Otherwise I'll try to implement this this week.

commonsguy, 2018-01-21:
That's possible. Do you want:

- A PR based on the current `master` (2.x API)?
- A PR based on the `mr/rxfingerprint_builder` branch (3.x API)?
- Just a pointer to a commit in a forked repo showing a possible 2.x implementation, that you integrate as you see fit?
- Wait until 3.x is released, then submit a PR off of the new `master`?
- Something else?

I'm happy to help, but you seem to be in the middle of a significant API change. IMHO, that means the timing of the suggestion might be good, but the timing of a PR might be... less good.

Mauin, 2018-01-21:
I think adding this to the `2.x` API still makes sense. It also makes this feature available to more users of the library in the v2 API compared to the potential smaller userbase of a v3.

Integrating these changes into the potential v3 API should not be too hard to do afterwards. Especially as the proposed changes for the v3 API are quite significant I am not sure about the timing of it yet and if they are worth it in the end.

So I would say adding the support for `char[]` based on current `master` and the v2 API is totally sensible.

Mauin, 2018-02-01:
Closing via #84

closed 2018-02-01 by Mauin