Mauin/RxFingerprint #85: NPE on Xiaomi MI 5s

PblryH, 2018-02-28:
`Fatal Exception: java.lang.NullPointerException: Attempt to invoke interface method 'java.lang.String java.lang.CharSequence.toString()' on a null object reference
       at com.mtramin.rxfingerprint.FingerprintObservable$2.onAuthenticationError(FingerprintObservable.java:89)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.sendErrorResult(FingerprintManager.java:848)
       at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage(FingerprintManager.java:813)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:163)
       at android.app.ActivityThread.main(ActivityThread.java:6321)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:880)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:770)`

PblryH, 2018-02-28:
My solution
`emitter.onError(new FingerprintAuthenticationException(errString != null ? errString : "Error code: " + errMsgId));`

And solution for potential NPE on line FingerprintObservable.java:100
`FingerprintObservable.this.onAuthenticationHelp(emitter, helpMsgId, helpString != null ? helpString.toString() : "Help code: " + helpMsgId);`

Mauin, 2018-03-07:
Interesting. This looks like Xiaomi devices might not supply a string in those callbacks at all under some conditions. Do you want to create a PR with your proposed fix for this?

PblryH, 2018-03-10:
>  Do you want to create a PR with your proposed fix for this?
it would be nice.