Mauin/RxFingerprint #88: javax.crypto.IllegalBlockSizeException on few devices

Farasy, 2018-03-13:
For some reason devices like ZTE V6 Plus (6.0) throw `javax.crypto.IllegalBlockSizeException` where others, mainly mainstream Samsung, Huawei, Sony work as expected.

`RxFingerprint.encrypt(EncryptionMethod.AES, context, "a", "xyz").subscribe()`

At least one fingerprint has been enrolled.
Any ideas?

Mauin, 2018-03-16:
Do you have a more detailed Stacktrace for the exception? Which version of RxFingerprint are you currently using?

Farasy, 2018-03-20:
Unfortunately not. It's 2.2.1.

Mauin, 2018-03-20:
Without much more details it's hard to guess what actually is the problem in this case. For AES the key size is set to `256` by default which should be valid. Might be that the implementation on the ZTE device is somehow incompatible with this.

Farasy, 2018-03-20:
Yes...
Let me take a peek into firebase. May there's more to work with.

Farasy, 2018-03-20:
`Exception javax.crypto.IllegalBlockSizeException:
android.security.keystore.AndroidKeyStoreCipherSpiBase.engineDoFinal (AndroidKeyStoreCipherSpiBase.java:519)
javax.crypto.Cipher.doFinal (Cipher.java:2056)
com.a.a.g.a (SourceFile:110)
com.a.a.i$2.onAuthenticationSucceeded (SourceFile:114)
android.hardware.fingerprint.FingerprintManager$MyHandler.sendAuthenticatedSucceeded (FingerprintManager.java:855)
android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage (FingerprintManager.java:803)
android.os.Handler.dispatchMessage (Handler.java:110)
android.os.Looper.loop (Looper.java:203)
android.app.ActivityThread.main (ActivityThread.java:6255)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1063)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:924)`

Farasy, 2018-03-21:
Another device facing the same problem: One Plus 5T
But only after creating new fingerprints from the android settings. Old fingerprints worked well with the lib.

Mauin, 2018-03-21:
Interesting. Unfortunately I have neither device at hand to test it. Also the Stacktrace is not being much more helpful.

Which encryption method are you using? Did you change this at some point?

Farasy, 2018-03-21:
Same for me :/
Untouched encryption method is AES. See my code above.

It's strange..
`

@Override
--
  | protected final byte[] engineDoFinal(byte[] input, int inputOffset, int inputLen)
  | throws IllegalBlockSizeException, BadPaddingException {
  | if (mCachedException != null) {
  | throw (IllegalBlockSizeException)
  | new IllegalBlockSizeException().initCause(mCachedException);
  | }
  |  
  | try {
  | ensureKeystoreOperationInitialized();
  | } catch (InvalidKeyException \| InvalidAlgorithmParameterException e) {
  | throw (IllegalBlockSizeException) new IllegalBlockSizeException().initCause(e);
  | }
  |  
  | byte[] output;
  | try {
  | flushAAD();
  | byte[] additionalEntropy =
  | KeyStoreCryptoOperationUtils.getRandomBytesToMixIntoKeystoreRng(
  | mRng, getAdditionalEntropyAmountForFinish());
  | output = mMainDataStreamer.doFinal(
  | input, inputOffset, inputLen,
  | null, // no signature involved
  | additionalEntropy);
  | } catch (KeyStoreException e) {
  | switch (e.getErrorCode()) {
  | case KeymasterDefs.KM_ERROR_INVALID_INPUT_LENGTH:
  | throw (IllegalBlockSizeException) new IllegalBlockSizeException().initCause(e);
  | case KeymasterDefs.KM_ERROR_INVALID_ARGUMENT:
  | throw (BadPaddingException) new BadPaddingException().initCause(e);
  | case KeymasterDefs.KM_ERROR_VERIFICATION_FAILED:
  | throw (AEADBadTagException) new AEADBadTagException().initCause(e);
  | default:
  | //line 519
  | throw (IllegalBlockSizeException) new IllegalBlockSizeException().initCause(e);
  | }
  | }
  |  
  | resetWhilePreservingInitState();
  | return output;
  | }
`

Farasy, 2018-04-03:
Hey Mauin, here's another Stacktrace from Android Vitals:

`java.lang.NullPointerException: 
  at com.mtramin.rxfingerprint.AesDecryptionObservable (SourceFile:95)
  at com.mtramin.rxfingerprint.FingerprintObservable$2.onAuthenticationSucceeded (SourceFile:105)
  at android.hardware.fingerprint.FingerprintManager$MyHandler.sendAuthenticatedSucceeded (FingerprintManager.java:1005)
  at android.hardware.fingerprint.FingerprintManager$MyHandler.handleMessage (FingerprintManager.java:926)
  at android.os.Handler.dispatchMessage (Handler.java:105)
  at android.os.Looper.loop (Looper.java:164)
  at android.app.ActivityThread.main (ActivityThread.java:6809)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.Zygote$MethodAndArgsCaller.run (Zygote.java:240)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:767)`

ddeath, 2018-12-03:
Just going around and have similar issue. Try it in android studio emulator on Nexus One with api 23. You will probably get the same error. This could help you with debugging

filipatbnp, 2023-10-09:
any update on this? I have same problem