Mauin/RxFingerprint #91: java.lang.ClassCastException

djdominoSVK, 2018-03-19:
I am not able to use AES encryption. (Device: Samsung S8, OS version: 8.0, library version: 2.2.1)

Usage:
```
    override fun encrypt(key: String, token: String): Observable<FingerprintScanResult> {
        return RxFingerprint.encrypt(EncryptionMethod.AES, context, key, token).map {
            fingerprintResultAdapter.toDomain(it)
        }
    }
```

Crash:
```
Caused by: java.lang.ClassCastException: android.security.keystore.AndroidKeyStoreRSAPrivateKey cannot be cast to javax.crypto.SecretKey
     at com.mtramin.rxfingerprint.AesCipherProvider.getKey(AesCipherProvider.java:60)
     at com.mtramin.rxfingerprint.AesCipherProvider.findOrCreateKey(AesCipherProvider.java:54)
     at com.mtramin.rxfingerprint.AesCipherProvider.cipherForEncryption(AesCipherProvider.java:75)
     at com.mtramin.rxfingerprint.CipherProvider.getCipherForEncryption(CipherProvider.java:84)
     at com.mtramin.rxfingerprint.AesEncryptionObservable.initCryptoObject(AesEncryptionObservable.java:83)
```

Am I missing something or it's really issue?

djdominoSVK, 2018-03-19:
I am really sorry, but problem was on my side.

At first I used RSA encryption, and then changed it to AES. Using same key ended up with mentioned exception.

closed 2018-03-19 by djdominoSVK

Mauin, 2018-03-19:
Glad that you figured it out! 👍