Mauin/RxFingerprint #95: RxFingerprint.keyInvalidated() requires SDK 23+, but what is the alternative to detecting key invalidation on SDK <23?

TylerMcCraw, 2018-04-06:
I've recently upgraded to the new RC (`3.0.0-RC-1`)

Because `RxFingerprint.keyInvalidated` was marked with an annotation `@RequiresApi(23)`, when detecting errors, I now have to write something like so:
```
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M && RxFingerprint.keyInvalidated(throwable)) {
    // The keys you wanted to use are invalidated because the user has turned off her/his
    // secure lock screen or changed the fingerprints stored on the device
    // You have to re-encrypt the data to access it
} else if (Build.VERSION.SDK_INT < Build.VERSION_CODES.M && throwable is InvalidKeyException) {
    // Is InvalidKeyException the right exception to catch here??
}
```

Could you write up some documentation on what to do in the case of a key being invalidated when the SDK is less than 23, since we can't use `RxFingerprint.keyInvalidated()`? 
This would help devs using this library know when to force users to re-authenticate when their device is pre-23.

Or maybe `RxFingerprint.keyInvalidated()` could be written to automatically handle pre-23 devices so that developers don't have to write this code themselves and keyInvalidated() could catch all states where the developer needs to force a user to re-authenticate. I'm not sure if this is possible.

Mauin, 2018-04-07:
I added the `@RequiresApi(23)` annotation to `RxFingerprint.keyInvalidated` as the `KeyPermanentlyInvalidatedException` that the method checks for was only added in SDK 23.

Keep in mind the whole Fingerprint API was only introduced in SDK 23 and thus checking for the `KeyPermanentlyInvalidatedException` doesn't make sense in the first place without the Fingerprint API present.

TylerMcCraw, 2018-04-10:
Ok. This makes sense.
I keep forgetting that it wasn't added until 23.
Thanks!

closed 2018-04-10 by TylerMcCraw