Mauin/RxFingerprint #96: launchFingerprintEnrollment without hasEnrolledFingerprints

Flo354, 2018-04-06:
Hi,

I am using the release 3.0.0-RC-1 and I have a question.
You just deprecated some methods including "hasEnrolledFingerprints". If I don't use "hasEnrolledFingerprints", how can I know if I need to use the method "launchFingerprintEnrollment" on Android P?

I tried to test "rxFingerprint.encrypt(KEY_NAME, stringToEncrypt)"... but the exception returned is of no use.

> java.security.InvalidAlgorithmParameterException: java.lang.IllegalStateException: At least one fingerprint must be enrolled to create keys requiring user authentication for every use
        at android.security.keystore.AndroidKeyStoreKeyGeneratorSpi.engineInit(AndroidKeyStoreKeyGeneratorSpi.java:254)
        at android.security.keystore.AndroidKeyStoreKeyGeneratorSpi$AES.engineInit(AndroidKeyStoreKeyGeneratorSpi.java:54)
        at javax.crypto.KeyGenerator.init(KeyGenerator.java:519)
        at javax.crypto.KeyGenerator.init(KeyGenerator.java:502)
        at com.mtramin.rxfingerprint.AesCipherProvider.createKey(AesCipherProvider.java:62)
        at com.mtramin.rxfingerprint.AesCipherProvider.findOrCreateKey(AesCipherProvider.java:52)
        at com.mtramin.rxfingerprint.AesCipherProvider.cipherForEncryption(AesCipherProvider.java:71)
        at com.mtramin.rxfingerprint.CipherProvider.getCipherForEncryption(CipherProvider.java:89)
        at com.mtramin.rxfingerprint.FingerprintDialogAesEncryptionObservable.initCryptoObject(FingerprintDialogAesEncryptionObservable.java:89)
        at com.mtramin.rxfingerprint.FingerprintDialogObservable.subscribe(FingerprintDialogObservable.java:70)

I was surprised to see that in your code, the method "isAvailable" does not checks for enrollments for Android P. Is it normal?

Thanks,

Mauin, 2018-04-07:
I also realized this shortly after I released RC-1. So this is a good question. The methods for `hasEnrolledFingerprints` and `isHardwareAvailable` were deprecated because they are part of `FingerprintManager`, which is also deprecated in Android P in favor of the `FingerprintDialog` implementation which doesn't have such methods.

Instead it seems like `FingerprintDialog` will emit error callbacks with `FINGERPRINT_ERROR_HW_NOT_PRESENT` or `FINGERPRINT_ERROR_NO_FINGERPRINTS` to indicate that this is the case.

I think this is another reason why the suggestion in #93 and #89 make sense. Especially in the context for Android P and these new errors it makes sense providing them to users so that it is possible to detect the issues and launch the fingerprint setup screen if necessary. Possibly also with helper methods for `isHwNotPresent` and `hasNoFingerprintsRegistered` on the `FingerprintAuthenticationException`.

Flo354, 2018-04-09:
Also, would it be possible to have a result when the user comes back from the enrollment activity?
This way, we could have more pertinent actions according to the result.

For instance: 
1) the user cancelled the action, we can do that
2) the user successfully registered, we do that
...

Mauin, 2018-04-10:
I haven't personally played around with this Intent yet, I just saw that it exists and decided it would make sense in RxFingerprint. If we can start the Intent with a result and then forward these result items to the caller I think it's definitely something we can consider adding.