Mauin/RxFingerprint #109: Is this library still maintained

Flo354, 2018-10-25:
Hi,

Is this library still maintained or will it not be updated anymore?

Thanks,

devmike01, 2020-05-09:
I recently submitted a pull request for migration to  Androidx and support for onscreen fingerprint, but it has not been approved. I'm going to wait till the month-end. If they don't merge it by then, I will move the library to a separate repository and continue managing it.

Flo354, 2020-05-09:
Hello @devmike01,

Thanks for your answer. For my own need, I just switched to the new androidx biometric library. It is what I need and it's an official library.

So I think, I can just close this issue.

P.S: you should not wait and directly manage the library on your own repo.

closed 2020-05-09 by Flo354

devmike01, 2020-05-09:
Good one.